gqrx-sdr/gqrx #1349: Linking bug when make project?

TurnOffNOD, 2024-04-20:
I never modify the source code, just git clone and make it.

However, when it starts linking, it report errors:

```
[ 98%] Linking CXX executable gqrx
/usr/bin/ld: warning: librtlsdr.so.2, needed by /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_xtal_freq'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_reset_buffer'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_device_usb_strings'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_tuner_if_gain'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_cancel_async'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_agc_mode'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_tuner_gain_mode'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_sample_rate'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_usb_strings'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_tuner_gains'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_tuner_gain'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_direct_sampling'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_xtal_freq'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_index_by_serial'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_freq_correction'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_offset_tuning'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_center_freq'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_tuner_gain'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_close'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_tuner_type'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_center_freq'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_open'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_read_async'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_freq_correction'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_sample_rate'

/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_set_bias_tee'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_device_count'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `rtlsdr_get_device_name'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/gqrx.dir/build.make:1222: src/gqrx] Error 1
make[1]: *** [CMakeFiles/Makefile2:235: src/CMakeFiles/gqrx.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

The last version didn't have these bugs. I have all the dependencies installed.

Can anyone check it?

argilo, 2024-04-20:
Hi there,

From the error message it appears you don't have the correct version of librtlsdr installed, which is needed by gr-osmosdr. Thus the problem lies outside Gqrx.

If you need further help with this, I would suggest asking in the user support group: https://groups.google.com/g/gqrx

closed 2024-04-20 by argilo

TurnOffNOD, 2024-06-08:
> Hi there,
> 
> From the error message it appears you don't have the correct version of librtlsdr installed, which is needed by gr-osmosdr. Thus the problem lies outside Gqrx.
> 
> If you need further help with this, I would suggest asking in the user support group: https://groups.google.com/g/gqrx

I'm using Archlinux, and install these packages with net mirror repository. Is it means that the problem lies in distro?