gqrx-sdr/gqrx #1353: macOS ARM Compatibility?

EVRiNOM, 2024-05-07:
Hi!
I apologize in advance if it's a dumb question or something, but can someone please make a native build for ARM processors on macOS?
I know I can use Rosetta but I don't want to install it, like to keep it "clean".
Thank you.

Axpelle, 2024-05-07:
The latest version runs very smoothly on M1/M2 chips. You won't even know Rosetta is activated.

EVRiNOM, 2024-05-07:
> The latest version runs very smoothly on M1/M2 chips. You won't even know Rosetta is activated.

I'm sure it does, it's not that. Consider me weird.

argilo, 2024-05-07:
There is already an issue open for this. Please see #1104.

closed 2024-05-07 by argilo

EVRiNOM, 2024-05-07:
> There is already an issue open for this. Please see #1104.

My bad, thank you.