gqrx-sdr/gqrx #1354: Crash when using GQRX to playback IQ files without a connected Rx

mhassell, 2024-05-18:
Context:
I use GQRX both for actively receiving signals and for playing back recorded IQ files (that were recorded through GQRX).

Reproduce:
Open GQRX without a receiver attached to the computer.  You will get an error "No device found with serial number...".  The device selection window pops up, and just hit enter.  The same error as before pops up.  Press enter and the GQRX gui loads. Then, go to Tools -> IQ Recorder and start playing an IQ file.  When the file starts playing, hit the "Play" button again to try to pause the playback.   The GUI crashes and an exception pops up that "gqrx exited with an exception. No device found with serial number ...."

willcode, 2024-05-18:
As a workaround, set the input to use IQ file /dev/zero.

vladisslav2011, 2024-05-18:
Or you may choose "default" device (soundcard input) if gr-osmosdr is compiled with SoapySDR support.
[image]