gqrx-sdr/gqrx #1356: Compile without audio

xaratustrah, 2024-06-11:
If audio is not needed, I was wondering if it is possible to compile `gqrx` without audio support? Maybe similar to https://github.com/gqrx-sdr/gqrx/issues/258#issuecomment-167447953 by passing an argument to `cmake` but this time in order to turn off all audio dependencies / features?

kgarrels, 2024-06-12:
I did the cmake approach, see here:
https://github.com/kgarrels/gqrx/commit/a06c6360347ef6253337c8efa09adb67b1e99434?diff=split&w=0

xaratustrah, 2024-06-14:
Thanks. I just cloned your version from [here](https://github.com/kgarrels/gqrx/tree/panadapter_v4), in the build directory I run::

```bash
cmake -DLINUX_AUDIO_BACKEND=Null-audio ..
```
I still get:

```bash
CMake Error at CMakeLists.txt:198 (message):
  Invalid audio backend: should be either Pulseaudio, Portaudio, Null-audio,
  or Gr-audio
```

maybe I should invoke it differently?

kgarrels, 2024-06-15:
I implemented this only for Darwin (macos).
I tried to fix it, but I cannot currently test it.

Please pull and try again.

PS: my fork has many other changes vs. master , maybe you should try to just apply that commit to master