gqrx-sdr/gqrx #1359: Feature request: show strongest signal freq in waterfall popup

OH1KK, 2024-06-21:
When I have wide bandwith on waterfall, it is hard to get exact frequency of strong signal. For example if I have 60MHz on screen, moving mouse cursor just a bit, frequency varies about 500kHz. It would be nice if gqrx would show frequency of stongest signal on waterfall popup near the cursor, for example distance +/- 10 pixels in addition to current info.

[image]

willcode, 2024-06-21:
You can click on the circle to tune to the peak. Turning averaging up gives you a more stable peak (for many signals).

OH1KK, 2024-06-21:
I am aware of that, but problem here is that clicking a circle is not so easy when there is a lot of overlapping bookmarks. Trying to hit dot easily get missed and one of existing bookmark is going to be selected.

willcode, 2024-06-22:
If max hold is on, the markers will stay. Right, it's hard to select things where the density is high.

The biggest difficulty with the feature you're looking for is that the FFT history is not kept. It's just a pixmap being scrolled down the screen.