gqrx-sdr/gqrx #1360: Incorrect RDS flags

KubaPro010, 2024-07-02:
Both on FM stereo and mono the RDS flags are wrong, it should be the value 9 (0b1001 indicating stereo and dynamic pty) however in gqrx the flags say `Music Mono stPTY` which is not correct as it should be `Music Stereo`, RDS Spy does say the correct flags so something is wrong here (for the future: i wont debug)

vladisslav2011, 2024-07-03:
Confirmed.
Thanks.
Possible fix: https://github.com/gqrx-sdr/gqrx/pull/1361

KubaPro010, 2024-07-03:
The fix looks correct, i did make a python RDS encoder and somewhat decoder ([librds](https://flerken.zapto.org:1115/kuba/librds)), the decoder part looks like this:
```python
match segment:
                case 0:
                    det.di_dpty = is_di
                case 1:
                    det.di_compressed = is_di
                case 2:
                    det.di_artificial_head = is_di
                case 3:
                    det.di_stereo = is_di
```
where is_di is the di bit and match is just a switch alternative in python

and here's the encoder part:
```python
( basic.b | int(ta) << 4 | int(ms) << 3 | ( ( di >> (3-segment) ) << 2) | segment ) & 0xFFFF,
```
Specifically `( ( di >> (3-segment) ) << 2)`

closed 2024-08-17 by argilo