gqrx-sdr/gqrx #1368: Filter width is 'too wide'

T-McBride, 2024-08-08:
Filter width appears too wide regardless of selection. Screenshots attached. Everything else works but audio is obviously static. 

Experienced with 2.17.5 app image and with 2.17.2 distro package, Gentoo linux, 6.6.38 kernel. SDR is hackrf with sample rate of 8M.

[image]

[image]

Thanks in advance.

argilo, 2024-08-08:
The filter widths look correct to me.

Keep in mind that the frequency listed at the top is in Hz. You're tuned very low. HackRF can't receive much below 1 MHz.

T-McBride, 2024-08-08:
Perhaps I'm missing something as WFM "consumes" the entire fm bandwidth and then some.
e.g. set to 91 fm but filter occupies ~32 to ~150 FM. 

[image]

If this is correct apologies for misunderstanding and please close. Just confused as can't get any FM reception in GQRX but have no issues in GNU radio. (?) 

TY

T-McBride, 2024-08-08:
Just re-read your point and realized my decimal error. My sincere apologies. Set to 91.000.00 etc no issues. Appreciate the quick reply and great software.

closed 2024-08-08 by T-McBride

argilo, 2024-08-08:
I'm glad to hear you sorted it out.