gqrx-sdr/gqrx #1369: Raspi CLI configuration error

n1moy, 2024-08-17:
Endeavor to build gqrx using a Raspi command line interface. Failed attempts have included and omited the 'export CXXFLAGS=-O2' command. 

$ git clone https://github.com/gqrx-sdr/gqrx.git gqrx.git
$ cd gqrx.git
$ mkdir build && cd build
$ export CXXFLAGS=-O2
$ cmake ..
$ make

Unsure how to proceed following the 'CMake Error' message. 

`thsk@op25:~/gqrx.git/build $ cmake ..
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:135 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/home/thsk/gqrx.git/build/CMakeFiles/CMakeOutput.log"`

Appreciate any help/direction provided. 

Thanks in advance,

argilo, 2024-08-17:
You are missing the required dependencies. Please follow the instructions here:

https://www.gqrx.dk/download/gqrx-sdr-for-the-raspberry-pi

closed 2024-08-17 by argilo

n1moy, 2024-08-18:
Follow up: Raspi CLI configuration error #1369 

Starting with a fresh & full Raspi OS image [64-bit debian Bookworm] on a Raspi 4B. 
While using the SSH PuTTY client and these commands: 

$ sudo apt update && sudo apt upgrade -y
$ sudo apt install -y cmake
$ sudo apt-get install -y gnuradio-dev gr-osmosdr qt6-base-dev qt6-svg-dev qt6-wayland libasound2-dev libjack-jackd2-dev portaudio19-dev libpulse-dev

Results in a roadblock with this error while 'Processng triggers' for the required dependencies. 

...
Processing triggers for shared-mime-info (2.2-1) ...
Processing triggers for mailcap (3.70+nmu1) ...
Processing triggers for fontconfig (2.14.1-4) ...
Errors were encountered while processing:
 xtrx-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1) 

The simpler solutions have not resolved anything. 
$ sudo dpkg --configure -a
$ sudo apt update
$ sudo apt autoremove
$ sudo apt clean

The package name is unknown to me. Uncertain how to proceed. 

Thanks again,

argilo, 2024-08-18:
Try `sudo apt purge xtrx-dkms`

n1moy, 2024-08-19:
2024.Aug.19

Thank you, efficient solution to the dpkg error! Duplicated the dependencies install without glitch, completed the build and gqrx is operational with a RTL-SDR blog V3 dongle. 

Desire to use a blog V4 using the modified Osmocom drivers. Using these commands: 

# purge previous driver
$ sudo apt purge ^librtlsdr
$ sudo rm -rvf /usr/lib/librtlsdr* /usr/include/rtl-sdr* /usr/local/lib/librtlsdr* /usr/local/include/rtl-sdr* /usr/local/include/rtl_* /usr/local/bin/rtl_*

# install drivers
$ sudo apt update && sudo apt upgrade -y
$ sudo apt install -y libusb-1.0-0-dev git pkg-config
$ git clone https://github.com/rtlsdrblog/rtl-sdr-blog
$ cd rtl-sdr-blog/
$ mkdir build && cd build
$ cmake ../ -DINSTALL_UDEV_RULES=ON
$ make
$ sudo make install
$ sudo cp ../rtl-sdr.rules /etc/udev/rules.d/
$ sudo ldconfig
$ echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf

Attempts to start gqrx now results in..

thsk@op25:~ $ gqrx
gqrx: error while loading shared libraries: libgnuradio-osmosdr.so.0.2.0: cannot open shared object file: No such file or directory

Thanks again for your help,

argilo, 2024-08-19:
In addition to rebuilding rtl-sdr from source, you would also need to purge the gr-osmosdr package and rebuild that from source, then rebuild gqrx.

n1moy, 2024-08-26:
2024.Aug.26

This procedure resulted in an error. Uncertain of the correct method. 

$ sudo apt update && sudo apt upgrade -y
$ sudo apt install -y cmake libusb-1.0-0-dev pkg-config
$ git clone https://github.com/rtlsdrblog/rtl-sdr-blog
$ cd rtl-sdr-blog/
$ mkdir build && cd build
$ cmake ../ -DINSTALL_UDEV_RULES=ON
$ make
$ sudo make install
$ sudo cp ../rtl-sdr.rules /etc/udev/rules.d/
$ sudo ldconfig
$ echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf
$ cd $HOME
$ sudo apt install -y gnuradio-dev qt6-base-dev qt6-svg-dev qt6-wayland libasound2-dev libjack-jackd2-dev portaudio19-dev libpulse-dev
$ git clone https://github.com/gqrx-sdr/gqrx.git
$ cd gqrx
$ mkdir build && cd build
$ cmake ..
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAVE_STDATOMIC
-- Performing Test HAVE_STDATOMIC - Success
-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libOpenGL.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1")
-- Checking for module 'gnuradio-osmosdr'
--   Package 'gnuradio-osmosdr', required by 'virtual:world', not found
CMake Error at /usr/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Gnuradio-osmosdr (missing: GNURADIO_OSMOSDR_LIBRARIES
  GNURADIO_OSMOSDR_INCLUDE_DIRS)
Call Stack (most recent call first):
  /usr/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindGnuradio-osmosdr.cmake:30 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:140 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/thsk/gqrx/build/CMakeFiles/CMakeOutput.log".

Thanks again,