gqrx-sdr/gqrx #1370: Follow-up to 'Raspi CLI configuration error' - Issue #1369

n1moy, 2024-08-29:
2024.Aug.29

Attempting to achieve a gqrx build that supports the RTL-SDR blog V4 on a Raspi 4B. My latest shot resulted a 'cmake' error during the gqrx build. Think I need help with Argilo's instruction to, "purge the gr-osmosdr package and rebuild that from source". 

Starting fresh using a full [64-bit debian Bookworm] Raspi OS image.  

$ sudo apt update && sudo apt upgrade -y

# Build RTL-SDR Osmocom driver
$ sudo apt install -y cmake libusb-1.0-0-dev pkg-config
$ git clone https://github.com/rtlsdrblog/rtl-sdr-blog
$ cd rtl-sdr-blog/
$ mkdir build && cd build
$ cmake ../ -DINSTALL_UDEV_RULES=ON
$ make
$ sudo make install
$ sudo cp ../rtl-sdr.rules /etc/udev/rules.d/
$ sudo ldconfig
$ echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf
$ cd $HOME

# gqrx dependencies without gr-osmosdr
$ sudo apt install -y gnuradio-dev qt6-base-dev qt6-svg-dev qt6-wayland libasound2-dev libjack-jackd2-dev portaudio19-dev libpulse-dev

# Build gqrx
$ git clone https://github.com/gqrx-sdr/gqrx.git
$ cd gqrx
$ mkdir build && cd build
$ cmake ..
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAVE_STDATOMIC
-- Performing Test HAVE_STDATOMIC - Success
-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libOpenGL.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1")
-- Checking for module 'gnuradio-osmosdr'
-- Package 'gnuradio-osmosdr', required by 'virtual:world', not found
CMake Error at /usr/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
Could NOT find Gnuradio-osmosdr (missing: GNURADIO_OSMOSDR_LIBRARIES
GNURADIO_OSMOSDR_INCLUDE_DIRS)
Call Stack (most recent call first):
/usr/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
cmake/Modules/FindGnuradio-osmosdr.cmake:30 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
CMakeLists.txt:140 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/thsk/gqrx/build/CMakeFiles/CMakeOutput.log".

Appreciate any help/direction provided.

Thanks in advance,

closed 2024-08-29 by argilo

n1moy, 2024-08-29:
2024.Aug.29

Completed the build without errors. The result is an installation that works with the blog V3, but not the blog V4. Below is the series of build commands, followed by the terminal exchange for the V4. 

Appreciate your help and direction.

Thanks again,

---

# Starting with a fresh image
$ sudo apt update && sudo apt upgrade -y

# Built RTL-SDR Osmocom driver
$ sudo apt install -y libusb-1.0-0-dev cmake pkg-config
$ git clone https://gitea.osmocom.org/sdr/rtl-sdr.git
$ cd rtl-sdr
$ mkdir build && cd build
$ cmake ../ -DINSTALL_UDEV_RULES=ON
$ make
$ sudo make install
$ sudo ldconfig
$ cd $HOME
$ echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf
$ cd $HOME

# Built gr-osmosdr:
$ sudo apt install -y gnuradio-dev
$ git clone https://gitea.osmocom.org/sdr/gr-osmosdr.git 
$ cd gr-osmosdr
$ mkdir build && cd build
$ cmake -DCMAKE_CXX_FLAGS=-Wno-register ..
$ make
$ sudo make install
$ sudo ldconfig
$ cd $HOME

# Built gqrx
$ sudo apt install -y qt6-base-dev qt6-svg-dev qt6-wayland libasound2-dev libjack-jackd2-dev portaudio19-dev libpulse-dev
$ git clone https://github.com/gqrx-sdr/gqrx.git
$ cd gqrx
$ mkdir build && cd build
$ cmake ..
$ make
$ sudo make install
$cd $HOME

# Optimized signal processing routines
$ volk_profile

---

thsk@op25:~ $ gqrx
QStandardPaths: wrong permissions on runtime directory /run/user/1000, 0770 instead of 0700
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.5.1
built-in source types: file rtl rtl_tcp rfspace redpitaya 
rx_nb_cc :info: set_min_output_buffer on block 10 to 32768
Resampling audio 96000 -> 48000
BandPlanFile is /home/thsk/.config/gqrx/bandplan.csv
BookmarksFile is /home/thsk/.config/gqrx/bookmarks.csv
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.5.1
built-in source types: file rtl rtl_tcp rfspace redpitaya 
Using device #0 RTLSDRBlog Blog V4 SN: 00000001
Detached kernel driver
Found Rafael Micro R828D tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!
[R82XX] PLL not locked!
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
Reattached kernel driver
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.5.1
built-in source types: file rtl rtl_tcp rfspace redpitaya 
Using device #0 RTLSDRBlog Blog V4 SN: 00000001
Detached kernel driver
Found Rafael Micro R828D tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!
[R82XX] PLL not locked!
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
Reattached kernel driver

argilo, 2024-08-29:
Can you post the output of this command?
```
ldd `which gqrx`
```

argilo, 2024-08-29:
I suspect the problem is that installing `gnuradio-dev` also pulls in (as a recommended dependency) `librtlsdr0`. I think you can avoid that by installing `gnuradio-dev` like so:
```
sudo apt install -y gnuradio-dev --no-install-recommends
```

n1moy, 2024-08-29:
thsk@op25:~ $ ldd `which gqrx`
        linux-vdso.so.1 (0x0000007f82d3a000)
        libQt6Network.so.6 => /lib/aarch64-linux-gnu/libQt6Network.so.6 (0x0000007f828c0000)
        libQt6SvgWidgets.so.6 => /lib/aarch64-linux-gnu/libQt6SvgWidgets.so.6 (0x0000007f82890000)
        libgnuradio-osmosdr.so.0.2.0 => /usr/local/lib/aarch64-linux-gnu/libgnuradio-osmosdr.so.0.2.0 (0x0000007f827f0000)
        libpulse.so.0 => /lib/aarch64-linux-gnu/libpulse.so.0 (0x0000007f82780000)
        libpulse-simple.so.0 => /lib/aarch64-linux-gnu/libpulse-simple.so.0 (0x0000007f82750000)
        libgnuradio-digital.so.3.10.5 => /lib/aarch64-linux-gnu/libgnuradio-digital.so.3.10.5 (0x0000007f825b0000)
        libgnuradio-network.so.3.10.5 => /lib/aarch64-linux-gnu/libgnuradio-network.so.3.10.5 (0x0000007f82510000)
        libQt6Widgets.so.6 => /lib/aarch64-linux-gnu/libQt6Widgets.so.6 (0x0000007f81de0000)
        libQt6Gui.so.6 => /lib/aarch64-linux-gnu/libQt6Gui.so.6 (0x0000007f816e0000)
        libQt6Core.so.6 => /lib/aarch64-linux-gnu/libQt6Core.so.6 (0x0000007f811f0000)
        libgnuradio-analog.so.3.10.5 => /lib/aarch64-linux-gnu/libgnuradio-analog.so.3.10.5 (0x0000007f81140000)
        libgnuradio-filter.so.3.10.5 => /lib/aarch64-linux-gnu/libgnuradio-filter.so.3.10.5 (0x0000007f81030000)
        libgnuradio-blocks.so.3.10.5 => /lib/aarch64-linux-gnu/libgnuradio-blocks.so.3.10.5 (0x0000007f80d40000)
        libgnuradio-fft.so.3.10.5 => /lib/aarch64-linux-gnu/libgnuradio-fft.so.3.10.5 (0x0000007f80ce0000)
        libgnuradio-runtime.so.3.10.5 => /lib/aarch64-linux-gnu/libgnuradio-runtime.so.3.10.5 (0x0000007f80b00000)
        libgnuradio-pmt.so.3.10.5 => /lib/aarch64-linux-gnu/libgnuradio-pmt.so.3.10.5 (0x0000007f80a80000)
        libvolk.so.2.5 => /lib/aarch64-linux-gnu/libvolk.so.2.5 (0x0000007f809a0000)
        libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007f80780000)
        libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f806e0000)
        libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f806a0000)
        libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f804f0000)
        libgssapi_krb5.so.2 => /lib/aarch64-linux-gnu/libgssapi_krb5.so.2 (0x0000007f80480000)
        libbrotlidec.so.1 => /lib/aarch64-linux-gnu/libbrotlidec.so.1 (0x0000007f80450000)
        libzstd.so.1 => /lib/aarch64-linux-gnu/libzstd.so.1 (0x0000007f80390000)
        libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007f80350000)
        libproxy.so.1 => /lib/aarch64-linux-gnu/libproxy.so.1 (0x0000007f80310000)
        /lib/ld-linux-aarch64.so.1 (0x0000007f82cfd000)
        libQt6Svg.so.6 => /lib/aarch64-linux-gnu/libQt6Svg.so.6 (0x0000007f80290000)
        libboost_chrono.so.1.74.0 => /lib/aarch64-linux-gnu/libboost_chrono.so.1.74.0 (0x0000007f80260000)
        librtlsdr.so.0 => /lib/aarch64-linux-gnu/librtlsdr.so.0 (0x0000007f80230000)
        libboost_thread.so.1.74.0 => /lib/aarch64-linux-gnu/libboost_thread.so.1.74.0 (0x0000007f801f0000)
        libpulsecommon-16.1.so => /usr/lib/aarch64-linux-gnu/pulseaudio/libpulsecommon-16.1.so (0x0000007f80150000)
        libdbus-1.so.3 => /lib/aarch64-linux-gnu/libdbus-1.so.3 (0x0000007f800d0000)
        libspdlog.so.1.10 => /lib/aarch64-linux-gnu/libspdlog.so.1.10 (0x0000007f80030000)
        libfmt.so.9 => /lib/aarch64-linux-gnu/libfmt.so.9 (0x0000007f7fff0000)
        libEGL.so.1 => /lib/aarch64-linux-gnu/libEGL.so.1 (0x0000007f7ffb0000)
        libfontconfig.so.1 => /lib/aarch64-linux-gnu/libfontconfig.so.1 (0x0000007f7ff40000)
        libX11.so.6 => /lib/aarch64-linux-gnu/libX11.so.6 (0x0000007f7fde0000)
        libglib-2.0.so.0 => /lib/aarch64-linux-gnu/libglib-2.0.so.0 (0x0000007f7fc80000)
        libQt6DBus.so.6 => /lib/aarch64-linux-gnu/libQt6DBus.so.6 (0x0000007f7fbb0000)
        libxkbcommon.so.0 => /lib/aarch64-linux-gnu/libxkbcommon.so.0 (0x0000007f7fb40000)
        libGLX.so.0 => /lib/aarch64-linux-gnu/libGLX.so.0 (0x0000007f7fae0000)
        libOpenGL.so.0 => /lib/aarch64-linux-gnu/libOpenGL.so.0 (0x0000007f7fa60000)
        libpng16.so.16 => /lib/aarch64-linux-gnu/libpng16.so.16 (0x0000007f7fa00000)
        libharfbuzz.so.0 => /lib/aarch64-linux-gnu/libharfbuzz.so.0 (0x0000007f7f8e0000)
        libmd4c.so.0 => /lib/aarch64-linux-gnu/libmd4c.so.0 (0x0000007f7f8c0000)
        libfreetype.so.6 => /lib/aarch64-linux-gnu/libfreetype.so.6 (0x0000007f7f7e0000)
        libicui18n.so.72 => /lib/aarch64-linux-gnu/libicui18n.so.72 (0x0000007f7f4a0000)
        libicuuc.so.72 => /lib/aarch64-linux-gnu/libicuuc.so.72 (0x0000007f7f280000)
        libdouble-conversion.so.3 => /lib/aarch64-linux-gnu/libdouble-conversion.so.3 (0x0000007f7f240000)
        libb2.so.1 => /lib/aarch64-linux-gnu/libb2.so.1 (0x0000007f7f220000)
        libpcre2-16.so.0 => /lib/aarch64-linux-gnu/libpcre2-16.so.0 (0x0000007f7f180000)
        libsndfile.so.1 => /lib/aarch64-linux-gnu/libsndfile.so.1 (0x0000007f7f0e0000)
        libfftw3f.so.3 => /lib/aarch64-linux-gnu/libfftw3f.so.3 (0x0000007f7efa0000)
        libfftw3f_threads.so.3 => /lib/aarch64-linux-gnu/libfftw3f_threads.so.3 (0x0000007f7ef70000)
        libboost_program_options.so.1.74.0 => /lib/aarch64-linux-gnu/libboost_program_options.so.1.74.0 (0x0000007f7eef0000)
        libgmp.so.10 => /lib/aarch64-linux-gnu/libgmp.so.10 (0x0000007f7ee50000)
        libunwind.so.8 => /lib/aarch64-linux-gnu/libunwind.so.8 (0x0000007f7ee00000)
[... cut ...]
        libXdmcp.so.6 => /lib/aarch64-linux-gnu/libXdmcp.so.6 (0x0000007f7bdd0000)
        libcap.so.2 => /lib/aarch64-linux-gnu/libcap.so.2 (0x0000007f7bda0000)
        libgcrypt.so.20 => /lib/aarch64-linux-gnu/libgcrypt.so.20 (0x0000007f7bc90000)
        liblz4.so.1 => /lib/aarch64-linux-gnu/liblz4.so.1 (0x0000007f7bc40000)
        libbsd.so.0 => /lib/aarch64-linux-gnu/libbsd.so.0 (0x0000007f7bc00000)
        libgpg-error.so.0 => /lib/aarch64-linux-gnu/libgpg-error.so.0 (0x0000007f7bbb0000)
        libmd.so.0 => /lib/aarch64-linux-gnu/libmd.so.0 (0x0000007f7bb80000)

argilo, 2024-08-29:
As I suspected, gr-osmosdr is linked to the system librtlsdr instead of the one you built from source. Replacing the command to install `gnuradio-dev` as I suggested above should fix the issue.

If you want to avoid starting from scratch, you can remove the librtlsdr0 package `sudo apt purge librtlsdr0`, then rebuild gr-osmosdr and gqrx.

argilo, 2024-08-30:
I'm glad you were able to get it working. This should all become much easier in a year or so when the next version of Debian & Raspberry Pi OS are released. At that time, you'll be able to install Gqrx from the package repository, and it will support V4 dongles.

One suggestion: if you add `-DDETACH_KERNEL_DRIVER=ON` to the `cmake` line for the RTL-SDR driver, then it is no longer necessary to blacklist the `dvb_usb_rtl28xxu` kernel module, since the RTL-SDR driver will take care of disabling and re-enabling it when the RTL-SDR is used as an SDR. The version of the driver that ships in the Debian package manager uses the `-DDETACH_KERNEL_DRIVER=ON` option.

n1moy, 2024-08-30:
The current command is:
$ cmake ../ -DINSTALL_UDEV_RULES=ON

Changing that to:
$ cmake ../ -DINSTALL_UDEV_RULES=ON -DDETACH_KERNEL_DRIVER=ON
Removing:
$ echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf

Is this correct ?

argilo, 2024-08-30:
Yes.

n1moy, 2024-08-30:
The blog V3 is working. Here is that terminal exchange:

thsk@thsk:~ $ gqrx
QStandardPaths: wrong permissions on runtime directory /run/user/1000, 0770 instead of 0700
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.5.1
built-in source types: file rtl rtl_tcp rfspace redpitaya 
rx_nb_cc :info: set_min_output_buffer on block 10 to 32768
Resampling audio 96000 -> 48000
BandPlanFile is /home/thsk/.config/gqrx/bandplan.csv
BookmarksFile is /home/thsk/.config/gqrx/bookmarks.csv
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.5.1
built-in source types: file rtl rtl_tcp rfspace redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!
[R82XX] PLL not locked!
Reattached kernel driver

[screenshot attached

[image]
]

argilo, 2024-08-30:
Did you reboot after running the installation steps, to pick up the newly installed udev rules?

n1moy, 2024-08-30:
Good call! Operational with blog V4. 
[screenshot attached]

Thanks again, 

[image]