gqrx-sdr/gqrx #1381: Red Pitaya ppm issue

DrSDR, 2024-10-12:
using the ethernet red pitaya 125-14,  it works fine for RX,  however when i change the ppm field , i do not see the signal shift in freq.
it seems the ppm field has no effect on the red pitaya
redpitaya=192.168.1.100:1001  using gqrx software.
thanks

argilo, 2024-10-12:
This appears to be a bug in gr-osmosdr. The `redpitaya_source_c::set_freq_corr` method updates an instance variable, but does not immediately retune the radio:

https://gitea.osmocom.org/sdr/gr-osmosdr/src/branch/master/lib/redpitaya/redpitaya_source_c.cc#L227-L232

I expect that an update to the PPM value will take effect the next time the frequency is adjusted.

DrSDR, 2024-10-12:
ok thank you, i will give that a try.
you are talking about the hardware center freq or the tuner offset freq ?

argilo, 2024-10-12:
Hardware center frequency.

DrSDR, 2024-10-12:
ok thanks, it worked, enter in ppm value,  then change hardware freq,  i saw the ppm correction get applied.  red pitaya is now tuned up.   thanks for the help.