gqrx-sdr/gqrx #1393: gr-osmosdr build CMake error

n1moy, 2024-11-06:
2024.Nov.06

Have an existing gqrx 2.15.8 installation on a Linux Mint machine. Would like to upgrade to the most current version with divers for a RTL-SDR V4 dongle. (2.17.5 ?)

Attempting to use the procedure from a previous issue. 

<https://github.com/gqrx-sdr/gqrx/issues/1370> 

Encountered a CMake error while building the gr-osmosdr gnuradio I/O block.

Thanks in advance,

$ sudo apt update && sudo apt upgrade -y
$ sudo apt install -y libusb-1.0-0-dev cmake pkg-config
$ git clone https://gitea.osmocom.org/sdr/rtl-sdr.git
$ cd rtl-sdr
$ mkdir build && cd build
$ cmake ../ -DINSTALL_UDEV_RULES=ON -DDETACH_KERNEL_DRIVER=ON
$ make
$ sudo make install
$ sudo ldconfig
$ cd $HOME
$ sudo apt install -y gnuradio-dev --no-install-recommends
$ git clone https://gitea.osmocom.org/sdr/gr-osmosdr.git
$ cd gr-osmosdr
$ mkdir build && cd build
$ cmake -DCMAKE_CXX_FLAGS=-Wno-register ..
.
.
.
-- Building for version: 0.2.0.0 / 0.2.0
-- Using install prefix: /usr/local
-- Configuring done
CMake Error at /usr/lib/cmake/pybind11/pybind11Tools.cmake:165 (add_library):
  Target "osmosdr_python" links to target "sndfile::sndfile" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrPybind.cmake:261 (pybind11_add_module)
  python/bindings/CMakeLists.txt:28 (GR_PYBIND_MAKE_OOT)

CMake Error at lib/CMakeLists.txt:51 (add_library):
  Target "gnuradio-osmosdr" links to target "sndfile::sndfile" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

argilo, 2024-11-06:
For help with installation, please use the Google group: https://groups.google.com/g/gqrx

From the error message, it appears you don't have the `libsndfile1-dev` package installed.

closed 2024-11-06 by argilo

n1moy, 2024-11-06:
Thanks! 

Avoiding a google groups account.