gqrx-sdr/gqrx #1397: Soapy Remote on Windows

srs4511351, 2024-11-12:
The latest Windows version of gqrx (2.17.5) does not work with Soapy Remote.
When I enter the device string:
driver=remote,remote=tcp://192.168.1.220:55132,remote:driver=sdrplay,soapy=0
I get the message SoapySDR::Device::make() no match
Please select another device.

That device string does work on gqrx 2.14.4 that comes with the PothosSDR Windows development environment.
It also works on the Linux version 2.17.5 that I compiled.

Is Soapy Remote included in the Windows release?

----Steve

argilo, 2024-11-12:
As noted in the release notes, only RTL-SDR and AirSpy devices are supported in the experimental Windows release. Soapy Remote is not available.

srs4511351, 2024-11-12:
Thanks. Will all of the devices be included in the future?
I was able to use the RTL-SDR Spectrum server with the SDRplay RSP TCP Server, but it reduces my SDRplay device to the frequency range of an RTL-SDR.
I'll use the old version for now...

argilo, 2024-11-12:
Gqrx is maintained by volunteers. Additional device support will be added to the Windows build if and when someone contributes it.

In the meantime, if you'd like support for many more devices on Windows, I'd suggest installing Gqrx using Radioconda: https://github.com/ryanvolz/radioconda

srs4511351, 2024-11-12:
Thanks! good idea. I tried it and it works.
I have also experimented with MSYS2 https://www.msys2.org/. You can compile applications from github, etc. so I can install things that aren't preconfigured.

srs4511351, 2024-11-13:
I am having trouble installing on MSYS2.
I'm stuck the gr-osmosdr cmake step
I do have pybind11 installed and the installer is requiring it.
mingw-w64-ucrt-x86_64-pybind11 2.13.6-2
```
CMake Error at /ucrt64/lib/cmake/pybind11/FindPythonLibsNew.cmake:163 (message):
  Python config failure:

Call Stack (most recent call first):
  /ucrt64/lib/cmake/pybind11/pybind11Tools.cmake:50 (find_package)
  /ucrt64/lib/cmake/pybind11/pybind11Common.cmake:228 (include)
  /ucrt64/lib/cmake/pybind11/pybind11Config.cmake:250 (include)
  CMakeLists.txt:185 (find_package)

-- Configuring incomplete, errors occurred!
```

argilo, 2024-11-13:
Gqrx does not need Python support, so you can build without it. Add `-DENABLE_PYTHON=OFF` to the cmake command.

srs4511351, 2024-11-14:
I tried compiling gqrx with -DENABLE_PYTHON=OFF in the cmake command, but it failed with this error
`Could NOT find Gnuradio-osmosdr (missing: GNURADIO_OSMOSDR_LIBRARIES`

The reason is that in my previous message, gr-osmosdr failed with a vague error about a pybind11 Python config failure.
I am thinking about starting the entire installation over.

I was able to compile NRSC5 on this system, though.

By the way, there is no gqrx compile instructions in the build steps page that you sent the link to earlier.

argilo, 2024-11-14:
Sorry, I meant that gr-osmosdr should be built with `-DENABLE_PYTHON=OFF`, not Gqrx.

srs4511351, 2024-11-14:
I finally got gr-osmosdr to build and install by adding -DENABLE_PYTHON=OFF.
I don't have Airspy installed yet.
I had to make this change: `-DCMAKE_INSTALL_PREFIX=${RUNNER_TEMP}/msys64/mingw64 \`

Building gqrx
For gqrx, adding -DENABLE_PYTHON=OFF allowed cmake to complete.
The build did not go well.

This is the procedure that I used. Is it valid?
```
cd ~/
git clone https://github.com/gqrx-sdr/gqrx.git gqrx
cd gqrx
cmake -S . -B build -DCMAKE_INSTALL_PREFIX=${RUNNER_TEMP}/mingw64 -DCMAKE_BUILD_TYPE=Release -DENABLE_PYTHON=OFF
cmake --build build
( would have been the next step of the previous had succeeded): cmake --install build
```
I have attached the cmake and build (cmake --build build) terminal output.
[cmake_log.txt](https://github.com/user-attachments/files/17744634/cmake_log.txt)
[build_log.txt](https://github.com/user-attachments/files/17744519/build_log.txt)

argilo, 2024-11-14:
The build steps I linked to above are used to produce automated builds, so I expect they will work if you use them on your machine.

srs4511351, 2024-11-20:
I looked into what I have to do to run the yml script that you linked.
Apparently, it is to be run on Docker.
My new laptop has Windows 11 Home and cannot run Hyper V, required by Docker. It would run WSL, but that won't produce Windows binaries.
My desktop computer has Windows 10 Pro, but its CPU does not support SLAT which is required for Hyper V.

Is there another way to run the yml script?

I was able to compile and install gqrx on MSYS2 MINGW64 with python support disabled.
I would prefer to have python working in gr-osmosdr, as other software may need it.
gqrx says `built-in source types: file rtl rtl_tcp soapy redpitaya`, but soapy does not work.
It pops up SoapySDR::Device::make() no match Please select another device
That's the same error as the gqrx installation from github.

Sorry for my rambling, but I'm not used to Docker or SDRs on Windows. I mostly use my Raspberry Pi.

argilo, 2024-11-20:
> Apparently, it is to be run on Docker.

Actually, it's a workflow configuration for GitHub Actions: https://docs.github.com/en/actions/writing-workflows/workflow-syntax-for-github-actions

> Is there another way to run the yml script?

It can't be executed directly. It provides a set of steps that GitHub Actions should execute. But you can copy & paste individual commands from the Windows portion into an MSYS2 terminal to get the same result.

> It pops up SoapySDR::Device::make() no match Please select another device

SoapySDR doesn't come with any drivers built in. Each one (e.g. [SoapyRemote](https://github.com/pothosware/SoapyRemote), [SoapyHackRF](https://github.com/pothosware/SoapyHackRF), etc. needs to be individually built and installed.

eabase, 2025-02-20:
@srs4511351 
@argilo 
I'm also looking to get the Python package working in Windows.
Any suggestions how to get this working? 

I'm currently trying to build using MSYS/MINGW64 in hope to get an idea how to get it working in native windows. But the MSYS python packages are not standard as they are not installed using pip, but pacman, and their availability are limited...