gqrx-sdr/gqrx #1402: Crash when accessing any child windows

parkerlreed, 2025-01-12:
Arch Linux (testing repos/KDE beta)

qt6-base 6.9.0beta1-1
gqrx-git 1:r2617.5debefb9-1
wayland 1.23.1-1
plasma-workspace 6.2.90-3

When clicking on any button with a child window, program instantly crashes with

```
[parker@parker-framework gqrx-git]$ gqrx
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio v3.11.0.0git-877-gc1bb9a6c
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya 
rx_nb_cc :info: set_min_output_buffer on block 12 to 32768
Resampling audio 96000 -> 48000
BandPlanFile is /home/parker/.config/gqrx/bandplan.csv
BookmarksFile is /home/parker/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 14.2.1 20240910; Boost_108600; UHD_4.7.0.0-0-unknown
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio v3.11.0.0git-877-gc1bb9a6c
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya 
Using device #0 RTLSDRBlog Blog V4 SN: 00000042
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
QWaylandShmBuffer: mmap failed (Invalid argument)
QPainter::begin: Paint device returned engine == 0, type: 3
error marshalling arguments for add_buffer (signature oi): null value passed for arg 0
Error marshalling request: Invalid argument
The Wayland connection experienced a fatal error: Invalid argument
```

This can lead to a soft block of loading the program if the last run crashed/invalid settings are detected, which forces the device settings window upon launch.

Main window runs fine and is operational.

vladisslav2011, 2025-01-13:
Hello.
It looks like this is not Gqrx bug, but Qt bug.
Related bugreport: https://bugreports.qt.io/browse/QTBUG-132727

parkerlreed, 2025-01-13:
Aha thank you! Indeed after posting this I ran across another program doing it, which is when I started looking for a QWt side report. Good to know..

argilo, 2025-01-14:
I'll close this off for now, since it does not appear to be a Gqrx issue. Let me know if that turns out not to be the case.

closed 2025-01-14 by argilo