gqrx-sdr/gqrx #1403: Feature Reqeust: Remote Control IQ-Data recording

Akira25, 2025-01-14:
When using the remote control feature of gqrx, we can control starting `AOS` and stoping `LOS` audio recordings. It would be great, having the same for IQ-recordings.

To simplyfy the interoperation with software like GPredict, this could be an option, which one to start on receiving the commands.

mhassell, 2025-01-20:
Hi, I am new to gqrx development but I was thinking of an idea like this as well.  So perhaps one could expand the interface to include two commands, eg AOI and LOI that will trigger recording of the IQ data in lieu of the audio.  I can mess with this a bit and see how far I get.

Akira25, 2025-01-20:
Sounds great! I highly appreciate that you are giving it a try.

Concerning the command, it is maybe better to stick with AOS and LOS and configure in gqrx, which behaviour those will trigger.

Implementing AOI and LOI would require to patch gpredict accordingly, to achieve this feature.

In addition, this would need to implement an option in gpredict, whether to emit AOS or AOI.

In my humble opinion that would be more effort, than having this option in gqrx directly.

mhassell, 2025-01-24:
Hey, so I added the AOIQ and LOIQ capability through the gqrx remote utilities and I have a [PR](https://github.com/gqrx-sdr/gqrx/pull/1405) open for that.  It uses new tokens (AOIQ, LOIQ) to capture the raw data when triggered.  I also made some changes in gpredict to interface well with the AOIQ and LOIQ commands.  There is a [PR](https://github.com/csete/gpredict/pull/371) open for that as well.  You are welcome to clone and build those branches and mess with them - please let me know if you find any issues or have ideas for improvements.  I think I am going to polish my GQRX work a bit to make the IQ recorder reflect that it is being used and not just recording silently (as it currently does with my changes).

argilo, 2025-02-07:
New remote control commands (`U IQRECORD 1` / `U IQRECORD 0` / `u IQRECORD`)  were added in #1142.

I don't think it would be a good idea to modify the behaviour of the existing `AOS` / `LOS` commands.

closed 2025-02-07 by argilo