gqrx-sdr/gqrx #1407: Direct sampling problem in 2.15 and higher versions

good-point-lab, 2025-01-27:
Gqrx crashes with an opaque error popup on 2.15.X and higher versions. Works as expected on 2.14.6.  OSX Big Sur 11.7.10.  Parameter set as direct_samp=2.  The same is true for values 0 or 1.  Please address!

argilo, 2025-01-29:
So far I haven't been able to reproduce this.

Are you using the DMG release? Could you share a copy of your `~/.config/gqrx/default.conf` file?

good-point-lab, 2025-01-29:
Hi Clayton and thanks for the reply!
Yes, I am using DMG releases for the versions.
The default.conf file is attached.
-Alex

argilo, 2025-01-29:
GitHub does not process email attachments, so I can't see your config file. You may have to copy & paste the contents.

good-point-lab, 2025-01-29:
Sure - here it is:

[General]
configversion=2
crashed=true

[audio]
gain=16
pandapter_max_db=-6
pandapter_min_db=-58
udp_host=localhost

[dxcluster]
DXCAddress=localhost
DXCFilter=
DXCPort=7300
DXCSpotTimeout=10
DXCUsername=nocall

[fft]
pandapter_max_db=-19
pandapter_min_db=-100

[gui]
***@***.***(\x1\xd9\xd0\xcb\0\x3\0\0\0\0\0\x9e\0\0\0\x32\0\0\x4U\0\0\x2\xe1\0\0\0\x9e\0\0\0N\0\0\x4U\0\0\x2\xe1\0\0\0\0\0\0\0\0\x5\0\0\0\0\x9e\0\0\0N\0\0\x4U\0\0\x2\xe1)
***@***.***(\0\0\0\xff\0\0\0\0\xfd\0\0\0\x2\0\0\0\x1\0\0\x1\"\0\0\x2M\xfc\x2\0\0\0\x2\xfc\0\0\0,\0\0\x1\x81\0\0\x1^\0\a\xff\xff\xfa\0\0\0\x1\x2\0\0\0\x3\xfb\0\0\0\x18\0\x44\0o\0\x63\0k\0I\0n\0p\0u\0t\0\x43\0t\0l\0\0\0\0\0\xff\xff\xff\xff\0\0\0\xf0\0\xff\xff\xff\xfb\0\0\0\x12\0\x44\0o\0\x63\0k\0R\0x\0O\0p\0t\x1\0\0\0\0\xff\xff\xff\xff\0\0\x1^\0\a\xff\xff\xfb\0\0\0\xe\0\x44\0o\0\x63\0k\0\x46\0\x66\0t\0\0\0\0\0\xff\xff\xff\xff\0\0\0\xc8\0\a\xff\xff\xfc\0\0\x1\xae\0\0\0\xcb\0\0\0\xc4\0\xff\xff\xff\xfa\0\0\0\0\x2\0\0\0\x2\xfb\0\0\0\x12\0\x44\0o\0\x63\0k\0\x41\0u\0\x64\0i\0o\x1\0\0\0\0\xff\xff\xff\xff\0\0\0\xc4\0\xff\xff\xff\xfb\0\0\0\xe\0\x44\0o\0\x63\0k\0R\0\x44\0S\0\0\0\0\0\xff\xff\xff\xff\0\0\0h\0\xff\xff\xff\0\0\0\x3\0\0\0\0\0\0\0\0\xfc\x1\0\0\0\x1\xfb\0\0\0\x1a\0\x44\0o\0\x63\0k\0\x42\0o\0o\0k\0m\0\x61\0r\0k\0s\0\0\0\0\0\xff\xff\xff\xff\0\0\x1\x42\0\xff\xff\xff\0\0\x2\x95\0\0\x2M\0\0\0\x1\0\0\0\x2\0\0\0\b\0\0\0\x2\xfc\0\0\0\x1\0\0\0\x2\0\0\0\x1\0\0\0\x16\0m\0\x61\0i\0n\0T\0o\0o\0l\0\x42\0\x61\0r\x1\0\0\0\0\xff\xff\xff\xff\0\0\0\0\0\0\0\0)"

[input]
bandwidth=500000
device="rtl=0,direct_samp=2"
frequency=7490300
***@***.***(\0\0\0\b\0\0\0\x1\0\0\0\x6\0L\0N\0\x41\0\0\0\x2\0\0\0\xf8)
sample_rate=1152000

[receiver]
demod=2
filter_high_cut=5000
filter_low_cut=-5000
offset=467900

[remote_control]
allowed_hosts=127.0.0.1

argilo, 2025-01-29:
Unfortunately I still can't reproduce the issue, so I'll have to leave this unless you can come up with more clues about what's going on. You're running rather old versions of macOS and Gqrx, so it may be worth updating if you can.

good-point-lab, 2025-01-29:
I understand, thank you. I will post more information on the open issue if
I collect more details. The Mac I am running GQRX on is older, and Big Sur
is the highest supported version. I have a new Mac, but it has an M1 chip.
It seems that GQRX does not yet support M1-based macOS, does it?
(I am building a Pandapter for a rig + SW monitoring, so direct dumpling is
a key functionality.)

argilo, 2025-01-29:
The latest version of Gqrx does support ARM-based Macs. There is a separate disk image for them.

good-point-lab, 2025-01-29:
Are there plans to support ARM with DMG distributions?  Could you please
send me a link to a stable disk image to try?

argilo, 2025-01-29:
https://github.com/gqrx-sdr/gqrx/releases

argilo, 2025-01-29:
You'll want `Gqrx-2.17.6-arm64.dmg`.

argilo, 2025-02-04:
Was the problem solved by moving to the current Gqrx version?

good-point-lab, 2025-02-04:
Sorry for the delay, surgery distracted me for some time.  Yes, the latest
DMG release runs on M1/15.3 OSX and the direct_samp parameters do work.
Thank you again.

argilo, 2025-02-04:
Thanks for confirming. In that case, I'll close off this issue.

closed 2025-02-04 by argilo