gqrx-sdr/gqrx #1410: Can not access remote control API from a local network's remote host

good-point-lab, 2025-02-10:
Here is all the information and I have tried several combinations without success.  My goal is to remotely control Gqrx running on RP from an ESP32 MCU using the TCP library.  From the MCU, Gqrx will be essentially on a remote host.  Please advise or address this issue.  I have found a few older discussions on line, however, the problem persists...

Allowed hosts
127.0.0.1
::ffff:10.0.0.213

Local host works as expected:
telnet localhost 7356
Trying ::1...
Connected to localhost.
Escape character is '^]'.
f
1072800

The remote host does not:
telnet 10.0.0.213 7356
Trying 10.0.0.213...
Connected to 10.0.0.213.
Escape character is '^]'.
Connection closed by foreign host.

Gqrx v2 17.6-16-g6191581 rtl=0,directsamp=2
Gqrx running on the latest OS 64-bit Raspberry Pi Bookworm hosted on Raspberry Pi B3

A remote attempt was done with Telnet from OSX Sequoya on M1

good-point-lab, 2025-02-10:
The actual config string is rtl=0,direct _samp=2.  The Gqrx itself in the direct sampling mode running just fine on RP B3.

argilo, 2025-02-10:
You need to add the IP address of the remote machine to the "Allowed hosts" list.

good-point-lab, 2025-02-11:
This is exactly what I did.  Added it to the list with the Gqrx menus and
verified after reboot.  No result ….

argilo, 2025-02-11:
It appears that you entered the IP address of the local host, not the remote host.

argilo, 2025-02-11:
Also, what version of Gqrx are you running? Starting in version 2.15.8 it is no longer necessary to prefix IPv4 addresses with `::ffff:`.

good-point-lab, 2025-02-11:
Thanks for your prompt responses!!
I included all the information in the issue description. Gqrx was built
from the code, and it is Gqrx v2 17.6-16-g6191581.
I have tried IP with and without  "::ffff:"
Still, Telnet from the host running Gqrx works, but from my laptop does not
work:

Trying 10.0.0.213...
Connected to 10.0.0.213.
Escape character is '^]'.
Connection closed by foreign host.

From the local host
telnet localhost 7356
Trying ::1...
Connected to localhost.
Escape character is '^]'.
f
1072800

argilo, 2025-02-11:
What is your laptop's IP address?

good-point-lab, 2025-02-11:
It is assigned by the Wi-Fi gateway to the  Raspberry Pi and it
is 10.0.0.213.  I can SSH to the PI without a problem.

argilo, 2025-02-11:
You need to enter your laptop's IP address into Gqrx, not the Raspberry Pi's IP address.

good-point-lab, 2025-02-11:
Oh boy, stupid me - let me try it right away. Thank you, Clayton!

good-point-lab, 2025-02-11:
It works like a charm, thanks again!

argilo, 2025-02-11:
Glad to hear it! I'll close off this issue, since the feature appears to be working as intended.

closed 2025-02-11 by argilo

good-point-lab, 2025-02-11:
Yes, please.  I should guess it myself... Lost focus at some point.