gqrx-sdr/gqrx #1414: Recent macOS builds fail to run

argilo, 2025-03-10:
Automated macOS builds currently fail to run, due to a signature validation error when loading libfftw3f.3.dylib. This library comes from Homebrew: https://formulae.brew.sh/formula/fftw

The problem began when the fftw bottle was updated to 3.3.10_2. `otool -l` shows that the library's SDK version is set to "n/a", but it needs to be set to 10.9 or later for the corresponding signature to be accepted.

Homebrew is tracking the issue here: https://github.com/Homebrew/homebrew-core/issues/209981

For now, I guess we'll have to wait until the issue is fixed and an updated bottle is published.

markjfine, 2025-04-01:
This problem might have gotten worse with macOS 15.4:

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: @executable_path/../libs/libfftw3f.3.dylib
Referenced from: <A68FDE49-617F-3C92-9C16-939F421776CF> /Applications/Gqrx.app/Contents/Libs/libgnuradio-fft.3.10.12.0.dylib
Reason: tried: '/Applications/Gqrx.app/Contents/libs/libfftw3f.3.dylib' (duplicate LC_RPATH '@executable_path/../libs/')
(terminated at launch; ignore backtrace)

Up to this point I've been able to build a macOS self-contained app, which worked until I upgraded to 15.4 this morning.

I should point out that this may be a dylibbundler issue, since gqrx can be run from the command like after doing a 'sudo make install'.

I'll continue to investigate.

precursorca, 2025-04-02:
Just updated to macOS 15.4 and same issue with Gqrx 2.17.6

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: @executable_path/../libs/libfftw3f.3.dylib
Referenced from: <9D01DFDC-930F-3D8D-896D-BABE294602F4> /Applications/Gqrx.app/Contents/libs/libgnuradio-fft.3.10.11.0.dylib
Reason: tried: '/Applications/Gqrx.app/Contents/libs/libfftw3f.3.dylib' (duplicate LC_RPATH '@executable_path/../libs/')
(terminated at launch; ignore backtrace)

markjfine, 2025-04-02:
Follow-up: Ran the app build script commenting out everything at the dylibbundler line and afterwards, forcing the app to use unmodified dependencies from their originally installed (via homebrew or otherwise) locations, then popped the resulting GQRX.app into /Applications. This app works - for now.

I suppose it will continue to work until one of the homebrew dependencies changes, in which case GQRX will likely need to be rebuilt. This is one of the reasons I like having all of the dependencies built with it copied into the app's Contents/libs folder. Then I only have to worry about occasional gnuradio changes that require gr-iqbal, libosmo-dsp, or gr-osmosdr  to be rebuilt (usually just the latter as the other two rarely change).

argilo, 2025-04-05:
Hmm, the 2.17.6 x86 DMG runs fine for me on macOS 15.4.

markjfine, 2025-04-05:
> Hmm, the 2.17.6 x86 DMG runs fine for me on macOS 15.4.

That's quite possible. gnuradio (3.10.12) was updated on 19 Feb 25 and libfftw3f.3.dylib was updated 08 Feb 25.
Since the dmg was built on 29 Nov 24 it has an older (and likely more compatible) version set embedded under the libs folder that it's running from, as pulled by dylibbundler at the time.

What's really weird is whatever dylibbundler is pulling into libs given the current build (2.17.6-26) is causing the runtime error. The current build's executable (Contents/MacOS/gqrx) will work fine with currently updated libraries if you skip pulling everything into libs using dylibbundler. It's almost as if dylibbundler is pulling in an alias file that's not linked properly.

precursorca, 2025-04-05:
I see the same behaviour.
2.17.6 x86 works.
2.17.6 ARM does not work (as above).
I am using an Apple Silicon Mac with 15.4.

markjfine, 2025-04-05:
So if I understand this correctly, the problem manifests itself due to dylibbuilder's (in)ability to correctly re-sign dylibs with an SDK version set to 'n/a'?

argilo, 2025-04-05:
I don't know what dylibbundler's involvement is, but the SDK version being set to "n/a" is definitely a problem, as macOS does not accept code signatures unless it is 10.9 or later.

markjfine, 2025-04-05:
When dylibbundler runs, it attempts to resign each library that is moved into Contents/lib in the resulting app cluster. This is when gqrx.app causes the resulting error above.

If dylibbundler is not run, no Contents/lib folder is created, and nothing is moved, re-referenced, nor re-signed. The executable in Contents/MacOS uses whatever libraries are installed (and signed) as they are. The resulting gqrx.app works fine this way.

Realize you're busy, but in case you're curious, attached is the piped output when dylibbundler is run on an M1 box, with additional terminal warnings added at the end -  lots of
`warning: changes being made to the file will invalidate the code signature in: Gqrx.app/Contents/libs/<dylib filename>`
when Xcode's install_name_tool is spawned from dylibbundler.

[dylibbuilder_output.txt](https://github.com/user-attachments/files/19616220/dylibbuilder_output.txt)

Now that I think of it, I should add that when I updated to macOS 15.4, I also updated Xcode and any associated console tools to 16.3, which may also be contributing to this.

markjfine, 2025-04-05:
Interesting. The x86 release ran on my M1 using Rosetta 2. Bounced around in the dock for a while but it ran and did not get the duplicate LC_RPATH error for fftw.

markjfine, 2025-04-10:
When I checked Homebrew earlier I read the thread more closely. Reading between the lines, this problem maybe much deeper than we thought. Last we saw they were testing a fix in early March, only to be marked as stale in early April.

I fully imagine at some point I'll be doing a brew update that has all of my 341 Sequoia-based ARM bottles to download and 'pour'... and hating it.

geerlingguy, 2025-04-14:
It looks like on Homebrew's side, the fix should be done in GCC upstream: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=119172

I'm running into this issue as well on my M4 Max.

markjfine, 2025-04-14:
> This is the Intel / x86 version, but it seems to run fine.

@geerlingguy 
It will run if you have Rosetta, but won't be as efficient as a native Silicon build.

precursorca, 2025-04-17:
FYI,
Arm version still crashes in the newly released macOS 15.4.1

markjfine, 2025-04-17:
> Arm version still crashes in the newly released macOS 15.4.1

I'd expect that, since the problem is the way Homebrew built fftw (and likely almost everything else) for ARM.

nisembedded, 2025-04-28:
FYI,
Still crashed on new Mac OS 15.4.1

maxbundscherer, 2025-05-10:
FYI
Same behavior here on Mac M1 with Mac OS 15.4.1

argilo, 2025-05-12:
Thanks for the information, but there's no need for further reports about the DMG releases failing to run. They won't begin working until either the upstream Homebrew issue is fixed, or we move away from Homebrew.

markjfine, 2025-05-12:
Was going to say...

nisembedded, 2025-05-13:
@argilo would you want to discuss anything related to migration? I have an experience with fix issues related to Mac Mx series, even for software from AD.

argilo, 2025-05-13:
Yes, if you have a proposal for how to fix this issue, I would be interested to hear it.

markjfine, 2025-05-13:
I am able to create Gqrx.app after compilation using the bundling script - if the sections to bundle (dylibbundler) and re-sign (codesign) the dylibs are commented out. However, this would presume that users of Gqrx.app have already installed those libraries using Homebrew using the same installation prefix you built gqrx against (i.e., /opt/homebrew/Cellar).

Wondering if MacPorts can be used as an alternative library source for building the app. I haven't checked to see if they have all of the required dylibs, nor if their implementations have the same build issues as Homebrew.

markjfine, 2025-05-13:
How hard would it be to re-baseline your build environment under Conda?

argilo, 2025-05-13:
I don't know.

I know little about software packaging, and would appreciate help.

kgarrels, 2025-05-13:
> markjfine

I am building for macos for some time, I can send the steps from a clean conda environment, probably tomorrow.

kai

kgarrels, 2025-05-14:
I am building with radioconda no homebrew or macports involved. This uses qt5:

- install radioconda with the graphical installer https://github.com/ryanvolz/radioconda/releases/tag/2025.03.14
- install a few dependecies:
```
conda init
conda activate base
conda install cmake rhash=1.4.5 libboost-devel
```

- go into gqrx source, cmake, build and run:
```
cd gqrx_conda/gqrx/build/
cmake ..
make
./src/gqrx
```

I will not finalize the bundling process, as I do not bundle, but use locally installed libs.
First steps could be:
- source for dylibbundler: https://github.com/auriamg/macdylibbundler
- initail attempts to adopt macos_bundle.sh, not finalized:
```
#!/bin/bash -e

GQRX_VERSION="$(<build/version.txt)"
IDENTITY=Y3GC27WZ4S
#BREW_PREFIX="$(brew --prefix)"
MACDEPLOYQT6="${CONDA_PREFIX}"/bin/macdeployqt

mkdir -p Gqrx.app/Contents/MacOS
mkdir -p Gqrx.app/Contents/Resources
mkdir -p Gqrx.app/Contents/soapy-modules

/bin/cat <<EOM >Gqrx.app/Contents/Info.plist
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple Computer//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
  <key>NSPrincipalClass</key>
  <string>NSApplication</string>
  <key>CFBundleExecutable</key>
  <string>gqrx</string>
  <key>CFBundleIdentifier</key>
  <string>dk.gqrx.gqrx</string>
  <key>CFBundleName</key>
  <string>Gqrx</string>
  <key>CFBundleIconFile</key>
  <string>gqrx.icns</string>
  <key>CFBundleShortVersionString</key>
  <string>$GQRX_VERSION</string>
  <key>CFBundleInfoDictionaryVersion</key>
  <string>6.0</string>
  <key>CFBundlePackageType</key>
  <string>APPL</string>
  <key>IFMajorVersion</key>
  <integer>1</integer>
  <key>IFMinorVersion</key>
  <integer>0</integer>
</dict>
</plist>
EOM

/bin/cat <<EOM >/tmp/Entitlements.plist
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple Computer//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
  <key>com.apple.security.cs.allow-unsigned-executable-memory</key>
  <true/>
</dict>
</plist>
EOM

cp build/src/gqrx Gqrx.app/Contents/MacOS
cp resources/icons/gqrx.icns Gqrx.app/Contents/Resources
# NOTE: PlutoSDR is built locally, so it will not in the brew path
cp "${CONDA_PREFIX}"/lib/SoapySDR/modules*/libremoteSupport.so Gqrx.app/Contents/soapy-modules
chmod 644 Gqrx.app/Contents/soapy-modules/*

#dylibbundler -s "${CONDA_PREFIX}"/opt/icu4c/lib/ -od -b -x Gqrx.app/Contents/MacOS/gqrx -x Gqrx.app/Contents/soapy-modules/libPlutoSDRSupport.so -x Gqrx.app/Contents/soapy-modules/libremoteSupport.so -d Gqrx.app/Contents/libs/
dylibbundler -s "${CONDA_PREFIX}"/lib -od -b -x Gqrx.app/Contents/MacOS/gqrx  -x Gqrx.app/Contents/soapy-modules/libremoteSupport.so -d Gqrx.app/Contents/libs/
"${MACDEPLOYQT6}" Gqrx.app -no-strip -always-overwrite # TODO: Remove macdeployqt workaround
if [ "$1" = "true" ]; then
    "${MACDEPLOYQT6}" Gqrx.app -no-strip -always-overwrite -sign-for-notarization="${IDENTITY}"
else
    "${MACDEPLOYQT6}" Gqrx.app -no-strip -always-overwrite
fi

for f in Gqrx.app/Contents/libs/*.dylib Gqrx.app/Contents/soapy-modules/*.so Gqrx.app/Contents/Frameworks/*.framework Gqrx.app/Contents/Frameworks/*.dylib Gqrx.app/Contents/MacOS/gqrx
do
    if [ "$1" = "true" ]; then
        codesign --force --verify --verbose --timestamp --options runtime --entitlements /tmp/Entitlements.plist --sign "${IDENTITY}" "$f"
    else
        codesign --remove-signature "$f"
    fi
done

```

markjfine, 2025-05-14:
Nice start!

kgarrels, 2025-05-15:
> Nice start!

see my new progress today, bundling s*cks, took me a full day to understand and fix
Can this be integrated now somehow?

kgarrels, 2025-05-16:
final update, I cannot move any further because e.g. I cannot notarize, I need help here.

- I did all changes for using micromambe in https://github.com/kgarrels/gqrx/tree/master_macos_conda
- github actions produce a macos artifact, it does not run, probably because it is not correctly signed
- ci for linux fails, qt6 is not found, I don't know why.

@markjfine , @argilo , do you want to take over from here?

argilo, 2025-05-16:
Nice! Thanks for all the work on this so far.

> do you want to take over from here?

Yeah, I should be able to sort out the signing part, at least. I'm away on a trip for the next week though, so it will have to wait a bit.

argilo, 2025-05-16:
> ci for linux fails, qt6 is not found, I don't know why.

I see that you've changed the default value of `FORCE_QT6` to `ON', which requires Qt 6 to be used. But Ubuntu 20.04 only has Qt 5.

That setting is only meant to be used to override the automatic Qt version selection, which first tries Qt 6 and then falls back to Qt 5.

argilo, 2025-05-16:
> github actions produce a macos artifact, it does not run, probably because it is not correctly signed

I pushed your branch to the main repository, which should allow signing & notarization to take place:

https://github.com/gqrx-sdr/gqrx/actions/runs/15071946650

argilo, 2025-05-16:
```
 ---codesign---
Failed to parse entitlements: AMFIUnserializeXML: duplicate dictionary key near line 10
```

It looks like that's because `<key>com.apple.security.cs.allow-unsigned-executable-memory</key>` appears twice in `macos_bundle.sh`.

argilo, 2025-05-16:
Looking over that list of entitlements, I wonder whether adding `com.apple.security.cs.disable-library-validation` might solve our immediate problem.

argilo, 2025-05-16:
I removed the duplicate from your branch: https://github.com/gqrx-sdr/gqrx/actions/runs/15072286510

Also I'm trying a build of master with `disable-library-validation` added: https://github.com/gqrx-sdr/gqrx/actions/runs/15072309553

argilo, 2025-05-16:
I don't have a Mac with me at the moment, so I'd appreciate feedback on the builds above.

precursorca, 2025-05-16:
This version Gqrx v2.17.6-31-g161b73b crashes on My M1 15.5 MacBook Pro.

Crash log below:

Translated Report (Full Report Below)
-------------------------------------

Process:               gqrx [40640]
Path:                  /private/var/folders/*/Gqrx.app/Contents/MacOS/gqrx
Identifier:            dk.gqrx.gqrx
Version:               v2.17.6-31-g161b73b (???)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               1025

Date/Time:             2025-05-16 11:38:54.4370 -0500
OS Version:            macOS 15.5 (24F74)
Report Version:        12
Anonymous UUID:        BEAB1B87-B081-CB53-366E-510A7358042D

Sleep/Wake UUID:       66C70BFA-6B21-44E7-A77B-EAB07AC0609F

Time Awake Since Boot: 84000 seconds
Time Since Wake:       4293 seconds

System Integrity Protection: enabled

Dyld Error Message:
  1

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: @executable_path/../libs/libfftw3f.3.dylib
Referenced from: <EF1B4F1A-E84B-374C-9B8A-C9C23A12EC7C> /private/var/folders/*/Gqrx.app/Contents/libs/libgnuradio-fft.3.10.12.0.dylib
Reason: tried: '/private/var/folders/cn/8g44zkd51k9dxp59hy8dqfq4000101/T/AppTranslocation/815C0313-039C-4A11-81C2-873F54A5D851/d/Gqrx.app/Contents/libs/libfftw3f.3.dylib' (duplicate LC_RPATH ***@***.***_path/../libs/')
(terminated at launch; ignore backtrace)

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000006   x1: 0x0000000000000001   x2: 0x000000016bde8f90   x3: 0x00000000000000e7
    x4: 0x000000016bde8b90   x5: 0x0000000000000000   x6: 0x0000000000000000   x7: 0x0000000000000000
    x8: 0x0000000000000020   x9: 0x1066f27d311400a6  x10: 0x000000000004cd30  x11: 0x0000000000020000
   x12: 0x0000000000013f70  x13: 0x0000000000013f60  x14: 0x0000000000000002  x15: 0x0000000000000000
   x16: 0x0000000000000209  x17: 0xcadd00016bde8868  x18: 0x0000000000000000  x19: 0x0000000000000000
   x20: 0x000000016bde8b90  x21: 0x00000000000000e7  x22: 0x000000016bde8f90  x23: 0x0000000000000001
   x24: 0x0000000000000006  x25: 0x000000019aea4000  x26: 0x000000001f070004  x27: 0x0000000000000000
   x28: 0x000000016bdeaec0   fp: 0x000000016bde8b60   lr: 0x000000019af2c648
    sp: 0x000000016bde8b20   pc: 0x000000019af1a620 cpsr: 0x80001000
   far: 0x0000000000000000  esr: 0x56000080  Address size fault

Binary Images:
       0x104014000 -        0x10413ffff dk.gqrx.gqrx (v2.17.6-31-g161b73b) <c8f437de-b6c2-3ebf-ac87-4652a277be9e> /private/var/folders/*/Gqrx.app/Contents/MacOS/gqrx
       0x104630000 -        0x104743fff org.qt-project.QtNetwork (6.9) <f6a1858f-aee7-34ed-867f-b04c65f29177> /private/var/folders/*/Gqrx.app/Contents/Frameworks/QtNetwork.framework/Versions/A/QtNetwork
       0x104250000 -        0x104257fff org.qt-project.QtSvgWidgets (6.9) <f2fef41c-277a-3255-8733-3d144b14b2ce> /private/var/folders/*/Gqrx.app/Contents/Frameworks/QtSvgWidgets.framework/Versions/A/QtSvgWidgets
       0x104434000 -        0x1044cbfff libgnuradio-osmosdr.0.2.0.0.dylib (*) <b341d141-3f5c-385e-88c1-af3f32a5ad70> /private/var/folders/*/Gqrx.app/Contents/libs/libgnuradio-osmosdr.0.2.0.0.dylib
       0x104290000 -        0x10429ffff libportaudio.2.dylib (*) <c612e3d9-7e9e-3275-8d96-9bbb84dc5e59> /private/var/folders/*/Gqrx.app/Contents/libs/libportaudio.2.dylib
       0x104830000 -        0x1048a7fff libgnuradio-digital.3.10.12.0.dylib (*) <b73194b8-8456-34f4-87bc-917ceccdf578> /private/var/folders/*/Gqrx.app/Contents/libs/libgnuradio-digital.3.10.12.0.dylib
       0x104334000 -        0x104363fff libgnuradio-network.3.10.12.0.dylib (*) <ff8f20e3-291c-30ce-b432-a21adc45a606> /private/var/folders/*/Gqrx.app/Contents/libs/libgnuradio-network.3.10.12.0.dylib
       0x104528000 -        0x10454bfff libgnuradio-audio.3.10.12.0.dylib (*) <24157d06-6839-3935-80be-a02e7da7caa2> /private/var/folders/*/Gqrx.app/Contents/libs/libgnuradio-audio.3.10.12.0.dylib
       0x104fbc000 -        0x105393fff org.qt-project.QtWidgets (6.9) <9f70323a-df5f-3c2f-8e0b-5886513a6d99> /private/var/folders/*/Gqrx.app/Contents/Frameworks/QtWidgets.framework/Versions/A/QtWidgets
       0x10493c000 -        0x104987fff org.qt-project.QtSvg (6.9) <8aff1b6c-7fcf-3961-bfa1-f642dbd1e822> /private/var/folders/*/Gqrx.app/Contents/Frameworks/QtSvg.framework/Versions/A/QtSvg
       0x105d54000 -        0x10618bfff org.qt-project.QtGui (6.9) <cda6be5d-710e-3861-8eb0-7676ea9e2950> /private/var/folders/*/Gqrx.app/Contents/Frameworks/QtGui.framework/Versions/A/QtGui
       0x105640000 -        0x1059fbfff org.qt-project.QtCore (6.9) <19b92468-8f1e-3883-a8ca-639618760c43> /private/var/folders/*/Gqrx.app/Contents/Frameworks/QtCore.framework/Versions/A/QtCore
       0x104570000 -        0x104593fff libgnuradio-analog.3.10.12.0.dylib (*) <1975cec5-f1b0-384e-aeec-6b61731a1488> /private/var/folders/*/Gqrx.app/Contents/libs/libgnuradio-analog.3.10.12.0.dylib
       0x104a7c000 -        0x104ac3fff libgnuradio-filter.3.10.12.0.dylib (*) <5aa83b08-6476-35a4-b49c-ff870754124d> /private/var/folders/*/Gqrx.app/Contents/libs/libgnuradio-filter.3.10.12.0.dylib
       0x104cc0000 -        0x104d57fff libgnuradio-blocks.3.10.12.0.dylib (*) <0a029eea-c517-39d7-8785-3338759ab3da> /private/var/folders/*/Gqrx.app/Contents/libs/libgnuradio-blocks.3.10.12.0.dylib
[... cut ...]
      "rolloutId" : "652eff3d1bce5442b8d753c9",
      "factorPackIds" : {

      },
      "deploymentId" : 240000009
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "d7ff00dd-1c5a-4dbb-85d6-7a936d671384",
      "experimentId" : "67f46bbc84eaf455d73c6098",
      "deploymentId" : 400000001
    },
    {
      "treatmentId" : "28060e10-d4e5-4163-aa2b-d8bd088d5cbf",
      "experimentId" : "6685b283afc7c17197d69eec",
      "deploymentId" : 400000012
    }
  ]
}
}

---
Alex Narvey
***@***.***

argilo, 2025-05-16:
Thanks for the feedback. It seems `disable-library-validation` didn't help, then.

kgarrels, 2025-05-16:
> I removed the duplicate from your branch: https://github.com/gqrx-sdr/gqrx/actions/runs/15072286510

This one launches sucessfully on my macbook air M2, macos 15.5.
Can test in detail tomorrow.

kgarrels, 2025-05-16:
> Also I'm trying a build of master with `disable-library-validation` added: https://github.com/gqrx-sdr/gqrx/actions/runs/15072309553

launches as well...

argilo, 2025-05-16:
> launches as well...

Interesting. It appears @precursorca ran the same build, so I wonder why it failed in that case.

kgarrels, 2025-05-16:
> > launches as well...
> 
> Interesting. It appears [@precursorca](https://github.com/precursorca) ran the same build, so I wonder why it failed in that case.

that dump still contains the .../libs/.. folder, I assume it was a wrong artifact?

argilo, 2025-05-16:
The `/libs/` folder is still present in the "master with disable-library-validation added" version.

argilo, 2025-05-16:
(And it appears @precursorca was running that version, given the git hash in the version number.)

kgarrels, 2025-05-16:
> The `/libs/` folder is still present in the "master with disable-library-validation added" version.

that version still has dylibbundler active

kgarrels, 2025-05-17:
I tested on my main machine mac mini m2 pro, macos 15.5:
- ```disable-library-validation``` crashes:
```
- ./Gqrx.app/Contents/MacOS/gqrx
dyld[37080]: Library not loaded: @executable_path/../libs/libfftw3f.3.dylib
  Referenced from: <EF1B4F1A-E84B-374C-9B8A-C9C23A12EC7C> /Users/kai/Downloads/Gqrx.app/Contents/libs/libgnuradio-fft.3.10.12.0.dylib
  Reason: tried: '/Users/kai/Downloads/Gqrx.app/Contents/libs/libfftw3f.3.dylib' (duplicate LC_RPATH '@executable_path/../libs/')
zsh: abort      ./Gqrx.app/Contents/MacOS/gqrx
```
dylibbundler creates a second set of libraries in /libs/, which  leads to the error message/crash
macdeployqt6 gathers everything that is needed in /Frameworks/
this is why I commented out dylibbundler in macos_bundle.sh

- ```master_macos_conda``` works for me

markjfine, 2025-05-17:
Yeah, so this was the original problem... It's not that dylibbundler creates a second set, it's what it does to the second set that macOS doesn't like. When you don't include the set in /lib/ you're basically running off what's already been installed by the package manager (conda, macports, homebrew, etc.). That's great if you have all the required packages installed. But not so much for regular, non-technical radio users.

So if you're using conda, it seems that using the conda distro of fftw might not be any better than the homebrew one when it comes to creating a bundled app.

argilo, 2025-05-17:
Thanks for the insights! I'll have a closer look at it once I'm back from my trip.

Switching to Conda will still be nice since it has a lot more radio packages than Homebrew, and it reduces the number of package sources we need to deal with. But it looks like we won't necessarily need to switch immediately.

markjfine, 2025-05-17:
Ok... So I don't have Qt6 because of DReaM, so I changed the 6 to a 5 to use the Qt5 version of macdeployqt.
macdeployqt puts everything in Frameworks, but crashes because they're not properly codesigned (I think dylibbundler already did that for you using your own code, so the codesign section of the script was redundant).

When I uncomment the codesign section of the script the resulting .app gives an error "The application Gqrx.app can't be opened.", likely because I'm not authorised to codesign with that identity and considered it corrupt. Also likely why I had it commented out when building locally.

I don't have time right now, but I'll try replacing the Identity code in the script with my own to see if that works.

kgarrels, 2025-05-17:
> Yeah, so this was the original problem... It's not that dylibbundler creates a second set, it's what it does to the second set that macOS doesn't like. When you don't include the set in /lib/ you're [...]

Could you help me understand this?
If I do a local build and then macos_bundle.sh, I get a /Frameworks/ folder that contains _everything_:
```
./macos_bundle.sh
/Users/kai/radioconda
/Users/kai/radioconda/lib/qt6/bin/macdeployqt6
---macdeployqt 2---
---codesign---
ls Gqrx.app/Contents/Frameworks
libFLAC.12.dylib			libdeflate.0.dylib			libgnuradio-pmt.3.10.12.dylib		libpng16.16.dylib
libLerc.4.dylib				libdouble-conversion.3.dylib		libgnuradio-runtime.3.10.12.dylib	libportaudio.dylib
libQt6Core.6.dylib			libffi.8.dylib				libgnuradio-uhd.3.10.12.dylib		librtlsdr.0.dylib
libQt6DBus.6.dylib			libfftw3f.3.dylib			libgobject-2.0.0.dylib			libsharpyuv.0.dylib
libQt6Gui.6.dylib			libfftw3f_threads.3.dylib		libhackrf.0.dylib			libsndfile.1.dylib
libQt6Network.6.dylib			libfmt.11.dylib				libiconv.2.dylib			libspdlog.1.15.dylib
libQt6Svg.6.dylib			libfreetype.6.dylib			libicudata.75.dylib			libssl.3.dylib
libQt6SvgWidgets.6.dylib		libgio-2.0.0.dylib			libicui18n.75.dylib			libthrift.0.21.0.dylib
libQt6Widgets.6.dylib			libglib-2.0.0.dylib			libicuuc.75.dylib			libtiff.6.dylib
libSoapySDR.0.8.dylib			libgmodule-2.0.0.dylib			libintl.8.dylib				libuhd.4.8.0.dylib
libairspy.0.dylib			libgmp.10.dylib				libjpeg.8.dylib				libusb-1.0.0.dylib
libairspyhf.0.dylib			libgmpxx.4.dylib			liblzma.5.dylib				libvolk.3.2.dylib
libbladeRF.2.dylib			libgnuradio-analog.3.10.12.dylib	libmirisdr.4.dylib			libvorbis.0.4.9.dylib
libboost_atomic.dylib			libgnuradio-audio.3.10.12.dylib		libmp3lame.0.dylib			libvorbisenc.2.0.12.dylib
libboost_chrono.dylib			libgnuradio-blocks.3.10.12.dylib	libmpg123.0.dylib			libwebp.7.dylib
libboost_filesystem.dylib		libgnuradio-digital.3.10.12.dylib	libogg.0.dylib				libwebpdemux.2.dylib
libboost_program_options.dylib		libgnuradio-fft.3.10.12.dylib		libopus.0.dylib				libwebpmux.3.dylib
libboost_serialization.dylib		libgnuradio-filter.3.10.12.dylib	liborc-0.4.0.dylib			libz.1.dylib
libboost_thread.dylib			libgnuradio-iqbalance.3.9.0.dylib	libosmodsp.0.dylib			libzstd.1.dylib
libc++.1.dylib				libgnuradio-network.3.10.12.dylib	libpcre2-16.0.dylib
libcrypto.3.dylib			libgnuradio-osmosdr.0.2.0.dylib		libpcre2-8.0.dylib
```
Isn't this what we want?
And why is not everything included when github actions build it?

markjfine, 2025-05-17:
So I think what's happening is that macdeployqt is moving things into /Frameworks/ instead of /libs/ but not telling all of the dylibs that's where they can be found... dylibbuilder does that. So when Gqrx.app is run, it finds libgnuradio-runtime, but libgnuradio-runtime is still looking for libgnuradio-pmt under the Homebrew prefix. Then you have a codesigning mismatch:
```
(code signature in <E9F548B5-10D0-3C2F-A45F-3D32CF8DEE52> '/opt/homebrew/Cellar/gnuradio/3.10.12.0_1/lib/libgnuradio-pmt.3.10.12.0.dylib' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs)
```

kgarrels, 2025-05-17:
ah that is maybe why I had to add the libpath to macdeployqt6:
```
if [ "$1" = "true" ]; then
    "${MACDEPLOYQT6}" Gqrx.app -no-strip -always-overwrite -sign-for-notarization="${IDENTITY}" -libpath=Gqrx.app/Contents/Frameworks
else
    "${MACDEPLOYQT6}" Gqrx.app -no-strip -always-overwrite -libpath=Gqrx.app/Contents/Frameworks
fi
```
still, in my local build, everything is there, and being found

markjfine, 2025-05-17:
So, finally got it to get past all the issues, but when running now pops up a an alert box with:
```
gqrx exited with an exception:
invalid msg port in connect() or disconnect()
More information can be found by launching it from the console
```

Runing from the console just gives:
```
built-in source types: file rtl rtl_tcp uhd rfspace airspyhf soapy redpitaya 
```
before the alert box shows.

But... it got past all the macOS nonsense even using Homebrew, so it's close.

Really need to finish this later.

markjfine, 2025-05-17:
I'm thinking this error is because it's signed differently than the one macOS previously saw and is now blocking things that Gqrx used to have permissions to do like "Files and Folders" & "Local Network".

I use the remote control feature and have it turned on, but I'm pretty sure that's not what this error message means. If it can't access files, it can't access the settings in ~/.config/gqrx and it will block remote access. It will also block USB communication to the SDR.

Pretty sure that revoking those old permissions and re-establishing them with the new app will fix that, but I'll have to test that theory another time.

markjfine, 2025-05-18:
I was able to reset Files and Folders using `sudo tccutil reset All dk.gqrx.gqrx`, but resetting Local Network requires dropping out of sip and trashing two plist files... not real comfortable doing that because that affects several other apps I run, including my Plex server.

That said, some research shows that the error message is generated by gnuradio. Others have gotten that error for various reasons, but I've not seen any solutions yet, either. It's still possible this is related to Local Network blocking USB comms to the SDR from gnuradio for this particular build.

Also had an inkling it might have had something to do with soapy-modules being taken out, but I re-added those portions back into the script (at least for `librtlsdrSupport.so`, which I think was previously needed) and I still get the error. Maybe it's expected to be in Frameworks as well.

Interesting problem.

kgarrels, 2025-05-18:
I made a few changes to my https://github.com/kgarrels/gqrx/blob/master_macos_conda/macos_bundle.sh:
- macdeployqt seems identical to macdeployqt6, it should now be "version agnostic"
- micromamba does not provide CONDA_PREFIX, so check for it, and use a hardcoded path if it is not there

For build.yaml:
- added all soapy-modules I found on my local system, soapy-modules, e.g. soapysdr-module-rtlsdr was missing

github actions now produce what they expect to produce, except for code signing.

I can also see the librtl problem, librtlsdr.0.dylib is included in /Framworks/, but librtlsdr.**2**.dylib seems to be requested, no idea where that comes from.

markjfine, 2025-05-18:
librtlsdr.dylib should be a link to librtlsdr.0.dylib and librtlsdr.0.dylib should link to librtlsdr.2.0.1.dylib (which on my system is curious bc it's supposed to be v2.0.**2** 🤷🏻‍♂️).

However, if you check the file size, it's possible that macdeployqt pulled in the appropriate dylib but renamed it to whatever the executable was linked with for packaging purposes. Both on my system are ~112kB.

kgarrels, 2025-05-18:
this is independant of gqrx bundling:
```
SoapySDRUtil --probe=driver=rtlsdr
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=rtlsdr
[ERROR] SoapySDR::loadModule(/Users/kai/radioconda/lib/SoapySDR/modules0.8/librtlsdrSupport.so)
  dlopen() failed: dlopen(/Users/kai/radioconda/lib/SoapySDR/modules0.8/librtlsdrSupport.so, 0x0001): Library not loaded: @rpath/librtlsdr.2.dylib
  Referenced from: <D19B0E5C-E821-3318-B4D4-7EE2263BA815> /Users/kai/radioconda/lib/SoapySDR/modules0.8/librtlsdrSupport.so
  Reason: tried: '/Users/kai/radioconda/lib/SoapySDR/modules0.8/../../librtlsdr.2.dylib' (no such file), '/Users/kai/radioconda/lib/SoapySDR/modules0.8/../../librtlsdr.2.dylib' (no such file), '/Users/kai/radioconda/lib/librtlsdr.2.dylib' (no such file), '/Users/kai/radioconda/bin/../lib/librtlsdr.2.dylib' (no such file), '/usr/local/lib/librtlsdr.2.dylib' (no such file), '/usr/lib/librtlsdr.2.dylib' (no such file, not in dyld cache)
Error probing device: SoapySDR::Device::make() no match
```

markjfine, 2025-05-18:
The part in this that puzzles me, is that both macdeployqt and dylibbundler check the link list within the application executable to figure out what libraries are required, then bundle them. The file listings are virtually identical, except for the filenames (and the addition of the requisite QT Framework only available through macdeployqt).

On the surface, the only differences are that dylibbunder appears to drag in the actual dylib and changes the link in the executable to the new path for that file, whereas macdeployqt drags in the actual dylib if it's not already in the bundle (whatever the path in the executable ultimately drills down to, once all the aliases are threaded) then renames it to whatever the filename in the executable is. Then macdeployqt changes the path in the executable, not the filename.

In both cases, there was no need to change anything within libfftw3f.3.dylib (nor libfftw3f_threads.3.dylib) - they are identical. Yet dylibbundler seems to somehow invalidate it within the security of macOS, whereas macdeployqt does not. That aside, running macdeployqt alone creates a different show-stopping issue somewhere within the bowels of gnuradio that I can't seem to put my thumb on.

If we solve this mystery, we've solved the problem.

kgarrels, 2025-05-18:
> invalid msg port in connect() or disconnect()

these errors occur if the gnuradio flowgraph is not correct, e.g. see https://lists.gnu.org/archive/html/discuss-gnuradio/2023-08/msg00110.html

[28 more replies]

PR #1424 (2025-05-27): Move macOS builds to Conda
Fixes #1414.

Recent Gqrx builds fail to start on macOS, due to a problem with Homebrew's fftw package. Building Gqrx using Homebrew packages has been troublesome, so I'd like to switch over to using Conda packages instead. A wide variety of SDR packages are available, which will eliminate the need to build dependencies from source. Gqrx's AppImage releases are already built using Conda packages, so we'll also have one fewer package source to deal with.

Many thanks to @kgarrels, who did the heavy lifting and prototyped a Conda-based build. I cleaned up the prototype [... cut]