gqrx-sdr/gqrx #1416: Recording or replay time length is wrong

stefanino-ch, 2025-03-21:
While recording a file (independent of the type) the length indicator counts per 10 seconds real time 1 second up. 

Or in other words, a recording of 60seconds will have a replay length of 6 seconds.

argilo, 2025-03-21:
I have not seen this behaviour before. Could you let me know how you reproduced this?

What version of Gqrx are you running? What SDR are you using? What is the "Device string" displayed in the "Configure I/O devices" panel? What is the input rate set to? Are you using decimation?

stefanino-ch, 2025-03-23:
I'm working on the current main tree. As of this week. 

Operating system current Fedora and current Ubuntu running both on Oracle Virtual Box. The target system will be Raspberry, but until now, I did all testing on virtualized Linux. I don't think this will be an issue. 

My SDR is a SDRPlay, on the Fedora system the SDRplay API 3.14 was used on the Fedora System latest and greatest 3.15 ist installed. 

All other packages needed to build gqrx are the lastest version ones. 

Setup as in the attached image gqrx-setup.jpg. 

[image]

How to reproduce: 
- start gqrx
- settings as in the image attached
- start DSP
- open I/Q recorder
- issue is with both recording formats, I used raw
- in the I/Q recorder window press rec
- observe the recording time
The image gqrx-recording.jpg shows my screen after ~60seconds of recording.

[image]

The image files.jpg shows the filesystem. From my point of view the filesize might be ok for a 60sec recording?

[image]

And here is what I got in the console where I started gqrx:

```
ste@ste-VirtualBox:~/sw-dev/gqrx-2025-03/gqrx/build/src$ ./gqrx
QSocketNotifier: Can only be used with threads started with QThread
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.9.2
built-in source types: file fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp xtrx 
rx_nb_cc :info: set_min_output_buffer on block 12 to 32768
Resampling audio 96000 -> 48000
BandPlanFile is /home/ste/.config/gqrx/bandplan.csv
BookmarksFile is /home/ste/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 13.2.0; Boost_108300; UHD_4.6.0.0+ds1-5.1ubuntu0.24.04.1
libusb: warning [libusb_exit] device 2.3 still referenced
libusb: warning [libusb_exit] device 2.1 still referenced
libusb: warning [libusb_exit] device 1.2 still referenced
libusb: warning [libusb_exit] device 1.1 still referenced
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.9.2
built-in source types: file fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp xtrx 
[INFO] devIdx: 0
[INFO] SerNo: 2105089645
[INFO] hwVer: 4
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0,000000
[INFO] Using format CF32.
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.9.2
built-in source types: file fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp xtrx 
[INFO] devIdx: 0
[INFO] SerNo: 2105089645
[INFO] hwVer: 4
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0,000000
[INFO] Using format CF32.
[WARNING] Not updating IFGR gain because AGC is enabled
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.9.2
built-in source types: file fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp xtrx 
[INFO] devIdx: 0
[INFO] SerNo: 2105089645
[INFO] hwVer: 4
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0,000000
[INFO] Using format CF32.
[WARNING] Not updating IFGR gain because AGC is enabled
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
rx_nb_cc :info: set_min_output_buffer on block 148 to 32768
Resampling audio 96000 -> 48000
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
```

argilo, 2025-03-23:
From the waterfall, it would appear that a lot of samples are being dropped. Do you hear choppy audio if you turn on a WFM demodulator?

stefanino-ch, 2025-03-24:
I wasn't get able to hear any audio during the tests. As in Switzerland all public radio emission on fm was stopped and replaced by DAB+ it is difficult to get any good reception. I will try to find a frequency gen for the generation of a test signal.

The time I've built last from scratch gqrx (around two years ago) I had the same issues. There I used the PR from @vladisslav2011 #1170 and #1139. Tried this as well this time, but during the long time the PRs was generated the code base seems to be too different now, to get meaningful results. At least for a short trial, I must maybe spend more time for a proper integration....

argilo, 2025-03-24:
OK. I'll close this off for now, as it appears the problem is with getting samples in via SoapySDR rather than with Gqrx itself.

closed 2025-03-24 by argilo

stefanino-ch, 2025-03-26:
Just to confirm, the problem was between chair and keyboard. The virtual machines I used don't bring the perfomance needed. After setting up everything on a raspberry 4 the same setup works without any issues. 
Sorry about this, and many thanks for your support and development effort.

argilo, 2025-03-26:
I'm glad to hear you tracked down the problem.

USB performance is a common problem when trying to run SDR applications in virtual machines.