gqrx-sdr/gqrx #1421: Using Device Scan when device is already connected

markjfine, 2025-04-21:
Found this while testing Airspy udev updates on macOS and Fedora 42, noticed some strange behaviour:

If you start the app and the last device you used is not found upon start up (e.g., it's not plugged in - I do this quite often, btw), it will tell you there's a (-1) error and give you an opportunity to scan for another device. Upon clicking Device Scan from within the I/O Devices dialog it will find it and you can move on. All as you would expect.

However, let's say you're already using an Airspy. It's plugged in, you're using it, and everything's fine. But you then stop it from playing and bring up the I/O Devices dialog again. When you click Device Scan... it comes up `Other...` and blanks the other fields out. You can click Cancel and then hit play and all will be fine, but it leaves the user to think the device is no longer connected when it really is.

This behaviour seems to indicate that the Device Scan function ignores finding the same device it thinks was already connected, instead of just reverting back to it.

Have been numerous changes lately: Fedora 42 (kernel 6.14.2-300.fc42.x86_64) macOS 15.4.1 (ARM), Qt 5.15.16_2 (on Mac) - but noticing this may just be coincidental.

argilo, 2025-04-26:
This could be an issue in gr-osmosdr, since it provides the device names. But I'll leave this open until the root cause is determined.

markjfine, 2025-04-27:
That's quite possible. Osmo sdr repos are apparently badly in need of maintainers. gr-osmosdr still can't find gr-iqbal properly if they come from builds separate from Gnuradio (e.g., mixed homebrew gnuradio and local gr-iqbal build on ARM), and that issue goes back to 2021 (though I just noticed the cause recently).